mcheah/Marlin4MPMD #4: Not flashing from SD card

mrpatg, 2018-07-30:
(ran into this initially, posting it in case anyone else does as well)

Formatted the SD card (using chromeos, so no options). Saved the appropriate files on the card and inserted into printer. After power-cycling the indicator light flashed white once, but otherwise went to normal operation. The main screen displayed nothing different, and no update was performed.

I suspect the issue is the wrong Filesystem/Blocksize/etc when formatting. Will verify and mark closed if this is the case.

mcheah, 2018-07-30:
I do not have a chrome book to test this on, but it's on my list to investigate ways to make this more reliable.

mcheah, 2018-07-31:
I did some testing today to help narrow down a lot of the common issues.  See here:
https://github.com/mcheah/Marlin4MPMD/wiki/SD-Cards
hopefully this helps.

mcheah, 2018-08-01:
@mrpatg,
Just following up, were you able to get things worked out?  I'm going to close this out in a couple days.

mrpatg, 2018-08-02:
@mcheah no unfortunately I couldn't get the printer to read the firmware from the card, even after formatting it correctly on windows. Firmware v37 and unsure if the card. Not a huge deal but perhaps I should try different cards.

mcheah, 2018-08-02:
does the SD card show .g files on the printer normally?  Some cards are flat out incompatible due to a hardware issue.
Can you confirm the following in windows:
Open a command prompt as administrator
type `diskpart`
type `list volume`
find the volume corresponding to your flash drive
type `select volume 3` (if 3 is the right volume for you)
then type `filesystems`
at the top you should see the following:
```
Current File System

  Type                 : FAT32
  Allocation Unit Size : 512
  Flags : 00000000
```
Additionally, if you open Disk Management, you should be able to see your disk at the bottom like this:
[image]

mrpatg, 2018-08-03:
No luck. Re-formatted, reloaded the files and tried again. Checked it in diskpart as well. I believe this is the stock 1gb card that came with the printer.

[image]

mcheah, 2018-08-03:
Hmm.. Interesting.  Thanks for including the video.  The short blip means it doesn't see any valid cards and just moves straight to the bootloader.  From the color screen UI, it suggests that you have an early version printer, which could have a different bootloader that doesn't support FAT32.  Do you know when you purchased the printer?  I do not have access to that hardware so it's difficult for me to troubleshoot remotely.  You may want to try formatting as FAT16 (shows up as FAT) in windows.  
You can print fine from this exact card in the normal application?  It looks like you have formatted the card correctly from windows, so we can take that out of the equation.  So the only steps I can recommend at this point are
- Make sure 100% that firmware.bin and fcupdate.flg are on the card
- Make sure the card is printable from the normal application, if you can't print from it, then it indicates the card is not compatible with the hardware
- Try formatting in FAT16/FAT32 in different allocation sizes 512/1024/1536/2048
- Try different cards that print in the normal application (I seem to find SDHC cards >2GB are less finicky with formatting), and then use them to upgrade the firmware.
- If none of the above are successful, and you're deadset on getting Marlin on there, you can buy an [st-link programmer](https://www.adafruit.com/product/2548), solder on some [debug headers](https://www.adafruit.com/product/3002) and reprogram the board through SWD.  If you end up going through this route, ping me on here, facebook or gitter and I'll talk you through what you need to do.  It's not particularly hard, just need a 4-pin header and download the [ST-Link Utility](https://www.st.com/en/development-tools/stsw-link004.html), but some people are averse to opening their printer and voiding their warranty.
Sorry you're having difficulties, I only have a limited set of hardware to work with.  I hope we can find a solution to help others with similar difficulties

mrpatg, 2018-08-03:
I will try all of this this afternoon and post results.

mrpatg, 2018-08-03:
FAT(16) - 16k - NO
FAT(16) - 32k - NO
FAT(16) - 64k - NO

FAT32 - 1024 - NO
FAT32 - 2048 - NO
FAT32 - 4096 - NO
FAT32 - 8192 - NO

Tossed in a gcode file to verify it could see it on the print menu and it could.

Id say for the time being just close this. I don't have any qualms about tearing the printer apart to manually program it, but I'll hold off for a bit. Maybe revisit in a month or two.

mcheah, 2018-08-03:
Okay, well thanks for being so thorough.  Sorry we can't get things working.  I'll leave this open for others.  My suspicion is that you have an early model printer that does something different due to a old version of the bootloader.  Can you confirm the approximate time you got the printer?

mrpatg, 2018-08-03:
Yeah order placed Nov 24 2017. If you have any other ideas to try let me know, I'll keep up with the repo going forward.

ApexProxy, 2019-07-28:
Just a heads up on SD compatibility if you haven't already resolved it, I purchased this 8GB Kingston based on recommendation and it has worked for everything from firmware to printing. For firmware I formatted it to Fat32 Allocation Size 512.

https://amzn.to/2FWmb74

mcheah, 2019-07-28:
Thanks, added it to the wiki.  Feel free to edit the wiki yourself if you find any updates