mcheah/Marlin4MPMD #5: No Serial Comms Octoprint

n9wxu, 2018-08-02:
I have flashed to v43+bin on a new delta.
The flash seemed to work and I can drive the printer from the front-panel and view the contents of the SD card.
However, when connecting via octoprint (linux pc) it goes to the connecting state for a bit and then aborts.  The start message never appears.

mcheah, 2018-08-02:
Did you turn off the malyan connection fix plugin?
can you try running `screen /dev/ttyACM0`
and then typing `G28` or `M503` into the terminal to see if things move around a bit?
type ctrl a + \ to quit screen

n9wxu, 2018-08-02:
I did kill malyan
I used miniterm /dev/ttyACM0 115200
G28 is homing it... just no start.

n9wxu, 2018-08-02:
It is showing v43.160.1 after flashing.
I can select the .bin in the print menu and it will reflash

mcheah, 2018-08-02:
v43 is the stock monoprice firmware... are you you installed the Marlin firmware?

n9wxu, 2018-08-02:
I grabbed the 10A v43+b  must be in the wrong place.

mcheah, 2018-08-02:
I didn't know you could issue a firmware update by the print menu... typically the procedure is to rename to firmware.bin on SD card, then power cycle

n9wxu, 2018-08-02:
I did that first.  I tried the print menu just to see what happened.

n9wxu, 2018-08-02:
OK I loaded the other 10A file, renamed it and it is working.  Homing is clearly a marlin style.

Why is there a 10A v43 option?

closed 2018-08-02 by n9wxu

mcheah, 2018-08-02:
the 5A/10A variants are based on what the power supply rating of your printer is.  I am not aware of Monoprice upgrading the power supplies to 10A, so if you still have the stock power supply you should use the 5A .bin file.  If you bought an aftermarket one then you can get the benefits of the 10A option