mcheah/Marlin4MPMD #10: Boot looping after changing thermistor type

harveyyau, 2018-08-13:
Hi,

My Mini Delta has an E3D clone hot end installed, with the ATC Semitec 100K thermistor, which corresponds to Type 5.  After pulling in the latest changes to the project   I had this working using previous builds of Marlin4MPMD.

I've gone back to a completely bare clone of the project except for one change.

-#define TEMP_SENSOR_0 75
+#define TEMP_SENSOR_0 5

Switching the value to 5 leaves the printer in an inconsistent state.  Switching it back to 75 allows the printer to boot.

mcheah, 2018-08-13:
I use a 12-bit ADC now, so all temperature tables need to be adjusted for 0-4095 codes.  If you add a *4 to every line in the thermistor table it should work correctly

harveyyau, 2018-08-13:
Thank you!  That fixed it for me.  I really appreciate this project, thanks for keeping me printing.

mcheah, 2018-08-13:
Feel free to submit a pull request and I'll merge in these changes.

closed 2018-08-13 by mcheah