mcheah/Marlin4MPMD #11: Extruder movement not working from LCD menu

McSwindler, 2018-09-08:
**Describe the bug**
When using the LCD menu to extrude or retract filament, it is using absolute positioning instead of relative. So you cannot continually extrude/retract.

**To Reproduce**
Steps to reproduce the behavior:
From the LCD menu, select move, then extruder. Attempt to extrude filament.
Sending a G91 command prior to using the LCD to extrude, will make it work as expected.

**Expected behavior**
The extruder should move relatively.

mcheah, 2018-09-08:
This is a known issue, see #20
https://github.com/mcheah/Marlin4MPMD/wiki/Frequently-Asked-Questions

emmanhello, 2019-03-06:
How do you insert/remove filament if the MPMD extruder is not moving in menu?

mcheah, 2019-03-06:
This firmware is really geared towards people who are using a usb connected tool like octoprint, pronterface, or repetier.  There was a nasty bug that was causing the print head to crash into the bed during prints, so the extruder menu was disabled.  This has been fixed in a beta update, but it's not a full release yet.

emmanhello, 2019-03-08:
Thanks for explication. For now, the solution would be to heat up nozzle and manually pull out the filament until beta is released. 
I think that's the only function needed to be able to use it without using USB.

mcheah, 2019-03-08:
You can check it out here:

https://github.com/mcheah/Marlin4MPMD/tree/experimental_G33?files=1

Just make sure you check the Readme for because there's a bunch of other changes added.  Most notably the towers are rotated in software to make the x/y plane more sensible, matching the tower swap mod in software.  I think it's m665 x120 y120 z120 to go back to normal

mcheah, 2019-04-21:
Fixed in 1.3.3

closed 2019-04-21 by mcheah