mcheah/Marlin4MPMD #12: clarify usage of M851

wtip, 2018-09-30:
Could you clarify the usage of M851 that you have documented here?https://github.com/mcheah/Marlin4MPMD/wiki/Calibration
`A nominal value between 0.3-0.5 is recommended. This is equivalent to calling M851 Z-<offset> (note this value is negative).`

However here https://reprap.org/wiki/G-code#M851:_Set_Z-Probe_Offset it mentions:

> This setting will be greater than zero on machines where the nozzle itself is used as the probe, pressing down on the bed to press a switch. (This is a common setup on delta machines.)

> Note that in Marlin 1.1.0 and later M851 sets the value literally as given, while Marlin 1.0.2 negates the absolute value.

So I would think that this should be a positive value. I had my printer set to `G0 Z0.4` when I started feeling resistance pulling paper between the nozzle and bed.

mcheah, 2018-09-30:
value should be negative.  The probe is effectively below the nozzle since the the switch will not register until it has sunk past the bed for 0.3-0.5mm or so.

wtip, 2018-09-30:
So is the information I quoted from the reprap wiki incorrect? In the main README you mention `Marlin4ST itself is based off of MarlinFW version 1.1.0-RC7`

mcheah, 2018-09-30:
maybe?  frankly it's a matter of a - sign.  Change M851 to several values, and then run G29 V4 P1 each time and you'll figure it out.

wtip, 2018-10-07:
For anyone else stumbling across this thread, unless I'm missing something it seems like it's more expedient to change the M851 value and then just run a `G29 P0 V4` followed by a `M503` to review the M665 settings.
My observation is that when using the `G29 P2` and `G29 P0` commands, the M851 offset only changes how the `M665 H` value is adjusted and does not change the offset on the recorded bed leveling matrix.

mcheah, 2018-10-07:
This is covered in the [faq](https://github.com/mcheah/Marlin4MPMD/wiki/Frequently-Asked-Questions)

wtip, 2018-10-07:
Which FAQ item covers adjusting `M851` ?

mcheah, 2018-10-07:
[Calibration](https://github.com/mcheah/Marlin4MPMD/wiki/Calibration)