mcheah/Marlin4MPMD #13: Writing to the chip's memory instead of an SD card

andybarry, 2018-11-19:
I have a situation where I will have more than one MPMD, which means that keeping SD cards with calibrations straight will be tough.

How hard do you think it would be to implement writing to the STM32F070CBT's memory instead?  I'd be happy to work on implementing it, but wanted to ask you (1) if you think its do-able and (2) where to start in the codebase to look at that.

mcheah, 2018-11-19:
It's definitely do-able, but it's not trivial either.  There's no EEPROM, so it has to be emulated by writing to flash ROM.  This requires using In-App-Programming of the flash, which is a separate peripheral driver.  It would also entail creating a data structure holding all of the saveable configuration parameters, which would have to remain consistent over time, etc etc.  In some ways I like the script solution because it's so simple and it's pretty much future proof.

The lowest hanging fruit in my opinion is instead of saving the entire configuration to flash ROM, you could instead store the machine serial number or a filename containing the config to load on powerup/m500/m501.  Then it will know to load M_CFG1, M_CFG2, etc.  This is still not ideal because you'd have to store the copy of all configs on all your SD cards, but it'd be a quick and dirty fix for your situation.

The code for writing out the settings is in `Config_StoreSettings` in `[Marlin\Configuration_store.cpp](https://github.com/mcheah/Marlin4MPMD/blob/faa9d260e574ec682558c0615b727e3ede66a779/MPMD_3dPrinter/Marlin/configuration_store.cpp#L207)`.  You'd need to update `Config_RetrieveSettings` too.  I'd strongly suggest starting with an example project from [stm32F0 cube](https://www.st.com/en/embedded-software/stm32cubef0.html) to test writing to flash before diving right in.  `STM32F070RB-Nucleo\Examples\FLASH\FLASH_EraseProgram` is probably the right example to start with.  I recommend the [NUCLEO-F070RB](https://www.digikey.com/products/en?keywords=stm32f070rb%20nucleo) as a development board for this kind of thing.

andybarry, 2018-11-19:
Thanks for the quick reply!  Dev board is on the way from Digikey.

It looks like ST has a [library](https://www.st.com/resource/en/application_note/cd00165693.pdf) that implements EEPROM via flash memory.  It seems like there might be a path where I could hook up that to Marlin's default EEPROM stuff and avoid having to build a separate configuration structure.  Is that reasonable or am I missing something?

mcheah, 2018-11-19:
It's possible.  Marlin EEPROM routines are completely out of date in this repo, so you'd have to add in all the extra parameters and settings that I've added to flash.

andybarry, 2018-11-19:
Another option might be just to write the script that is currently going to the SD card into flash and then read it automatically before every print.  Not the most space efficient I suppose -- how much room in the flash is there available?

andybarry, 2018-12-16:
To follow up on this, I managed to make the stock firmware work well enough for my application that it doesn't make sense for me to implement this right now.

closed 2018-12-16 by mcheah