mcheah/Marlin4MPMD #20: Marlin 2.0?

RedlineGT, 2019-02-12:
Please forgive me if this is inappropriate or not in the right spot.

Have you come across this?
https://github.com/MarlinFirmware/Marlin/issues/10739

It talks about how to compile Marlin 2.0 for the stm32f0. I know its a different board but maybe this may point you to good direction maybe?
Sorry if i am over my head. Just trying to contribute somehow.

N

mcheah, 2019-02-12:
Yes i have.  @xC0000005 has made some efforts on making a port work on the mpmd, but at this point I've diverged enough from the Marlin repo with custom ST-drivers that it may not make sense to merge the two.  I chose the philosophy of "compile specifically for ST" as opposed to "get Arduino to play nice with ST".

RedlineGT, 2019-02-12:
> Yes i have. @xC0000005 has made some efforts on making a port work on the mpmd, but at this point I've diverged enough from the Marlin repo with custom ST-drivers that it may not make sense to merge [...]

Fair enough, so will you use your ST-drivers to see if we can get Marlin 2.0 working for us peasants? I guess my point is to ultimately have as many new features ported for us. Unless you have and I missed it. I am not looking for any specific features. Just hoping somehow your hardwork can be used to easily integrate ongoing Marlin developments.
@xC0000005 how has things progressed on your end? Just curious

Thank you for the quick reply. I am forever thankful for all the work you have poured into this machine for us.

mcheah, 2019-02-13:
Unfortunately, it's quite a bit of work to port all of the marlin 2.0 features over as they've refactored the code base significantly.  I'd be more than happy to advise an effort for a marlin 2.0 port given my familiarity of the platform.  If there's a specific feature or issue you'd like to see addressed I can consider it, but I am also beginning to run into memory constraints of the stm32f0 processor.  Feel free to try the 1.3.1 release candidate in the experimental_g33 branch and see if it addresses any of your issues.  I'm looking for feedback before putting out a new release.

closed 2019-03-08 by mcheah