mcheah/Marlin4MPMD #21: MAXTEMP_110 is actually 109

emmanhello, 2019-03-12:
This isn't really an issue, I'm really sorry for opening a bug for just one degree celsius, if you think it is not worth the time, just close it. 

**Describe the bug**
If the bed temp reach exactly 110ºC, system reboots (security feature when maxtemp is exceed I suppose).
Probably the same issue for MAXTEMP_90 for 5A power supply

**To Reproduce**
With Pronterface, set the bed temperature above 109 through USB. Have a 10A psu.

**Expected behavior**
Should be able to reach and maintain MAXTEMP, and reboot only if it is exceeded, not equaled.

**Additional context**
Setting temp to 109 is not a problem at all, but is maxtemp is said to be 110, some people (like me) might think something in the hardware went wrong because that temp is too high for the Delta Mini. Also people likes round number.
Again, for 1ºC, this isn't really an issue...

mcheah, 2019-03-12:
Feel free to submit a pull request to fix.
TBH, if you're able to reach 109C, I'm happy to raise maxtemp higher, really the number one thing you want to catch is if somebody swapped the hotend/bed plugs.

emmanhello, 2019-03-13:
I will give it a try. What IDE are you using?

mcheah, 2019-03-13:
I use [openstm32](http://www.openstm32.org/HomePage)
There's some details on building here:
https://github.com/mcheah/Marlin4MPMD/wiki/Build

closed 2019-04-21 by mcheah