mcheah/Marlin4MPMD #22: Dies in middle of print

brazda, 2019-03-22:
**Describe the bug**
The printer resets in middle of the print. It seems no matter how I try. I have tried different slicers like cura and slic3r and I have tried printing from SD Card and Octoprint. I have a 10 amp power supply and I have tired both the 10 amp and the 5 amp versions and it seems to do the same.  

**To Reproduce**
Steps to reproduce the behavior:
1. Which g-codes can cause this issue?
any
2. Are you printing from USB or the SD card?
tried both
3. Are you using the LCD display to start prints or are you using commands from the USB?
tied both
4. What OS are you using?
Linux/Windows 
5. What software are you using?(Cura, Pronterface, Octoprint, Repetier...)
Cura, Slic3r, Pronterface, Octoprint
6. Did this work on the stock Malyan firmware?
yes, I re-flashed it back to stock and it seems to work 
7. What error message is reported?  Are the LED lights blinkly quickly
The LCD screen flashes and looks like a soft reboot as it does not show the bootloader loading but after it goes black for a second it loads to the default load screen
8. Post the full-code and status log, if you can.

I did get this from Octoprint logs

There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor
Connecting to: /dev/ttyACM0
Using Malyan/Monoprice Connection Fix Plugin to create serial connection
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x70275ff0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=15.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: Error:No Line Number with checksum, Last Line: 0
Recv: echo:Unknown command: "echo:Marlin 1.2.2-MPMD"
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Send: N1 M115*39
Recv: ok
Send: N2 M21*18
[...]
Recv: ok
Recv: Error:openRoot failed
Recv: code=3
Recv: \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00echo:SD init fail

**Expected behavior**
A clear and concise description of what you expected to happen.
not crash :)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
Linux Arch
Windows 10

 - Software: [Cura, Pronterface, Octoprint, Repetier...]
 - Version [e.g. 22]
Cura 3.6, Cura 4.0
lastest Pronterface
lastest Octoprint
latest slic3r

**Additional context**
Add any other context about the problem here.

mcheah, 2019-03-22:
So the only times I've seen a full reboot have been due to overloading the power supply.  Here are two things you can try to help narrow down the issue:

1.  Try a print without setting the bed heater on.  This will significantly reduce the load on the power supply.

2.  Try a print with the stock power supply and the 5A firmware.  This will tell for sure whether or not it is your replacement power supply.

What revision printer do you have, eg. what was the original firmware version?

Can you link me the actual g-code you are printing with?  This will tell me a lot more information like what temperatures you're printing at etc.

Can you provide a screenshot of the octoprint/pronterface temperature chart?  Sometimes there are some signs of what's going on with the printer

Have you upgraded the hotend or cooler?  Any additional mods like LED's or a larger heatbed?

brazda, 2019-03-29:
I will run these test and provide more info in a few days. I ended up moving back to the stock firmware as I needed the printer to print a part for my printer I am building.

I can tell you its not the power supply because it did the same with the 5amp firmware on both the 5 amp and the 10 amp power supply and the 10 amp firmware on the 10 amp ps.
As for original firmware I didn't right it down and no longer remember however my printer was purchased in January this year. 
As for gcode I noticed if you use the start gcode for cura on the mpmd wiki the printer will not start do to the save in the calibration code at the top. So I had to remove it.  When I tried my temp settings were 190 hotend and 60 bed i also did 240/80 as I was trying to print petg at first and move to pla to get working.

closed 2019-04-21 by mcheah