mcheah/Marlin4MPMD #23: Strange bug involving file (auto00.g) from stock card and booting

jacobalberty, 2019-03-27:
**Describe the bug**
It seems having both m_cfg.g and auto00.g on the sd card causes the firmware to stop working

**To Reproduce**
Back up your stock card, format it to fat32 with 512 byte sectors then copy the files from the back up onto it. install m4mpmd, execute m500 and then restart the printer. Octoprint will not be able to connect and none of the lcd menus work correctly.

If I eject the card then octoprint can connect again. if I remove m_cfg.g it works but I lose the stored settings. If I remove auto00.g it seems to work correctly.
Relevant log:

```
Trying baudrate: 115200
Recv: echo:SD card ok
Send: N0 M110 N0*125
Recv: echo:enqueueing "M23 auto0.g"
Recv: echo:enqueueing "M24"
Recv: echo:Now fresh file: m_cfg.g
Recv: File opened: m_cfg.g Size: 2049
Recv: File selected
Recv: echo:Now doing file: auto0.g
Recv: open failed, File: auto0.g.
See octoprint.log for details
Connection closed, closing down monitor
```

**Expected behavior**
I expect it to just work.

**Additional context**
I'm not sure more than removing auto00.g is needed, though a proper fix would be correcting the m_cfg.g loader. I do have other .gco files on the card (no other files ending in .g). So maybe a documentation update warning against having auto00.g on the card? it may even be all files ending in .g

mcheah, 2019-03-27:
This may be true, there's logic for auto.g but I've never tested it.  For now I recommend just copying your auto.g settings into m_cfg.g and it'll run on startup.  Unfortunately they'll be lost when issuing M500.

You're also welcome to try experimental_g33 branch which uses flash EEPROM and doesn't use m_cfg.g

mcheah, 2019-03-27:
Also don't embed M501 inside a gcode file.  Nested scripts are not supported.