mcheah/Marlin4MPMD #24: Home when

mojocorp, 2019-04-14:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

mojocorp, 2019-04-14:
Thank you so much for this amazing work ! It's working perfectly so far, way better than the original firmware. You just saved me a lot of pain. Thank you again!

mcheah, 2019-04-14:
Yup, this is a known issue.  Last time I tried it broke things.  If you're running octoprint you can add a on-cancel g-code script that will do this for now

Krishpannze, 2019-11-01:
I am using cura what should i do