mcheah/Marlin4MPMD #25: Move the hotend up (20mm) when 'Pause' is triggered from the UI

mojocorp, 2019-04-14:
**Is your feature request related to a problem? Please describe.**
When Pausing using the UI button 'Pause', the hotend just stop where it was. This lead to 
a melt mark but prevent inserting a nut for example.

**Describe the solution you'd like**
Moving the hotend 20-50mm up should solve this problem. The original firmware does this.

realbrianosaurus, 2019-08-07:
Actually, if you're implementing pause, can you do Advanced Pause and enable M600 Filament change :).  I've been using "M600 X55 Y0 Z15 L0" to swap filaments on my stock mini delta. The stock one just doesn't seem to understand how to pause over a glass bed.

mcheah, 2019-08-07:
Please make a separate issue if you're asking for additional functionality, it makes it easier to track

realbrianosaurus, 2019-08-10:
Will do. I figured it was related to pause changes, so I was just mentioning it here.