mcheah/Marlin4MPMD #26: Where to adjust Amp limit?

Grzy7316, 2019-04-21:
Where would I go to adjust the Amp limit? I have   I do not have a 10A power supply, but I do have a 7 amp power supply that I am using now (belonged to my MP Select mini). I would like to be able to maximize the performance I get from the 7A power supply without buying a new 10A power supply.

mcheah, 2019-04-21:
It's an all or nothing thing, either both heaters are on at the same time or only one of them.  You can try the 10A firmware, but there's still a chance your supply will trigger over currrent protection and reboot the system.

Grzy7316, 2019-04-29:
@mcheah Thank you for the information.  I Acquired a 10A power supply, should make things simpler

closed 2019-04-29 by Grzy7316