mcheah/Marlin4MPMD #27: Documentation does not describe what the FastSPI firmware is for.

alficles, 2019-05-02:
**Describe the bug**
There are several options for firmware, the distinctions between which are documented clearly in the install instructions. But the FastSPI firmware isn't mentioned anywhere I can find.

**Expected behavior**
It would be helpful to know what the FastSPI choice is used for.

mcheah, 2019-05-03:
There's a 10k pull down on one of the SD interface lines.  Running fast spi allows running at 16x faster comm speeds, potentially decreasing transfer times for high quality SD cards.  Because it requires altering the hardware with a soldering iron and potentially breaking the bootloader, I intentionally don't describe it to prevent idiots from bricking their printers and blaming me.  If you're willing to take the risk, PM me on gitter, otherwise don't worry about it.

closed 2019-05-03 by mcheah

alficles, 2019-05-03:
That's fine. I'm happy with it the way it is, I just didn't know that one was one to stay away from. I almost picked that FW because I assumed it was for the latest model. Maybe a comment warning folks about it? Caution is definitely advised here; I see why you kept it under the radar.