mcheah/Marlin4MPMD #28: Temperature too high

ehansen2580, 2019-05-08:
Tried running the autotune for PID and received an error message saying the temperature is too high.  It stopped the autotune and did not disconnect.

**To Reproduce**
Steps to reproduce the behavior:
1. Which g-codes can cause this issue?
M303 S<TEMP> C<CYCLES>

2. Are you printing from USB or the SD card?
USB through octoprint on a Pi

3. Are you using the LCD display to start prints or are you using commands from the USB?
commands through USB

4. What OS are you using?
most recent Octopi

5. What software are you using?(Cura, Pronterface, Octoprint, Repetier...)
Cura for slicing.  Octoprint for printing.

6. Did this work on the stock Malyan firmware?
I did not try the autotune with the Malyan software.  But I can print OK with it.

7. What error message is reported?  Are the LED lights blinkly quickly
Send: M105
Recv: ok T:26.6 /0.0 (3888) B:22.6 /0.0 (3920) @:0 B@:0
Send: M303 S<TEMP> C<CYCLES>
Recv: PID Autotune start
Recv:  T:26.6 /0.0 (3888) B:22.7 /0.0 (3919) @:127 B@:0
Recv:  T:28.2 /0.0 (3875) B:22.9 /0.0 (3918) @:127 B@:0
Recv:  T:30.8 /0.0 (3852) B:22.7 /0.0 (3919) @:0 B@:0
Recv:  T:34.3 /0.0 (3816) B:22.8 /0.0 (3919) @:0 B@:0
Recv: PID Autotune failed! Temperature too high
Recv: ok
Send: M105
Recv: ok T:39.9 /0.0 (3750) B:22.6 /0.0 (3920) @:0 B@:0
8. Post the full-code and status log, if you can.
Send: M105
Recv: ok T:26.6 /0.0 (3888) B:22.6 /0.0 (3920) @:0 B@:0
Send: M303 S<TEMP> C<CYCLES>
Recv: PID Autotune start
Recv:  T:26.6 /0.0 (3888) B:22.7 /0.0 (3919) @:127 B@:0
Recv:  T:28.2 /0.0 (3875) B:22.9 /0.0 (3918) @:127 B@:0
Recv:  T:30.8 /0.0 (3852) B:22.7 /0.0 (3919) @:0 B@:0
Recv:  T:34.3 /0.0 (3816) B:22.8 /0.0 (3919) @:0 B@:0
Recv: PID Autotune failed! Temperature too high
Recv: ok
Send: M105
Recv: ok T:39.9 /0.0 (3750) B:22.6 /0.0 (3920) @:0 B@:0

**Expected behavior**
A clear and concise description of what you expected to happen.
I thought it would run through a PID autotune and report back the proper settings to then send with the a command for memory storage.

mcheah, 2019-05-09:
Typically you have to set a temp and number of cycles when doing autotune.  Try M303 S150 C5

https://www.reprap.org/wiki/G-code#M303:_Run_PID_tuning

ehansen2580, 2019-05-09:
Yes, this fixed the problem I had.  Thank you.

closed 2019-05-09 by ehansen2580