mcheah/Marlin4MPMD #29: 1.3.3 on v44 printer: Steps per mm wrong, probably?

rcpoison, 2019-05-11:
**Describe the bug**
Bed leveling touches edges of print bed and print bed fixtures. 
Prints go outside of print bed.

**To Reproduce**
Steps to reproduce the behavior:

Install 1.3.3 ' ' ' firmware_5ALimit_v43+.bin``` on printer that came with v44 firmware.
Attempt print.

1. Which g-codes can cause this issue?
G29, ...

2. Are you printing from USB or the SD card?
SD Card

3. Are you using the LCD display to start prints or are you using commands from the USB?
LCD

4. What OS are you using?
Linux

5. What software are you using?(Cura, Pronterface, Octoprint, Repetier...)
Cura

6. Did this work on the stock Malyan firmware?
Yes

7. What error message is reported?  Are the LED lights blinkly quickly
None, but printer audibly communicated its agony.

8. Post the full-code and status log, if you can.

**Expected behavior**
Not this.

**Desktop (please complete the following information):**
 - OS: Linux
 - Software: Cura
 - Version 4.0.0

**Additional context**

After the first print attempt I flashed ```firmware_5ALimit.bin``` which resulted in the same behaviour.

gcode doesn't contain anything funky like M92.
Attached here: 
[MMD_caseVoltUpTop.gcode.gz](https://github.com/mcheah/Marlin4MPMD/files/3169521/MMD_caseVoltUpTop.gcode.gz)

mcheah, 2019-05-12:
Post your M503 settings.
You likely have mis-adjusted M665 R/L settings which will cause the printer to move farther than it thinks it is.  Print a 20mm calibration cube to confirm

rcpoison, 2019-05-12:
@mcheah My bad, I actually ran an endstop calibration script before flashing - forgot about that, and it seems it issued M500's >.<
Sorry for wasting your time :(

closed 2019-05-12 by rcpoison

mcheah, 2019-05-12:
Hey no problem.  Happy to help.