mcheah/Marlin4MPMD #30: SD-Card not accessible

THE-BOW, 2019-05-29:
I installed the Firmware from Stock to 1.3.3 10A Version
Now the MicroSD that i used to Flash is no more accessible.
It was read before without problems...

Terminal gives me back following:
Error:openRoot failed
[ERROR] Error:openRoot failed
code=3
echo:SD init fail

mcheah, 2019-05-30:
Did you install_fasSPI.bin?
What brand / capacity is your card?
If you installed with the SD card there should be no problem reflashing using the same card with the original firmware.  The bootloader and Marlin use a different SD card driver.
Also read this for proper formatting of SD cards.
If you are still having issues, pm me on gitter, Facebook or reddit and I'll walk you through your options to reflash without SD.

THE-BOW, 2019-05-31:
Hello mcheah,

i installed the provided firmware_10ALimit_v43+.bin

My SD is a Trancend 2GB (TS2GUSD)
I used the card to Flash the 1.3.3
After updating the firmware i got this error when i tried to access it.
Also reformating the SD in any format did not help... (I tried all options from FAT over FAT32 to exFAT) 

My printer is now back working
After trying a view different non branded SD`s i found a 64MB Nokia branded SD wich was recognized by the bootloader.
Flashing the stock firmware allowed me to access the Trancend SD again.

After getting back to stock i flashed 1.3.3 wich brought me the same issue again.

I willl keep the 100 year old SD for future rollbacks...

Many thanks

closed 2019-05-31 by THE-BOW