mcheah/Marlin4MPMD #31: Home action doesn't work after initial startup

Serneum, 2019-06-14:
**Describe the bug**
When initially turning on the MPMD, the Home Axes button does not do anything. I can hear a little bit of a mechanical movement, but it ends quickly. If I try to take the Home Axes action again, it works just fine. Any Home action after that one will also work. I believe this also happens the first time I hit Home Axes after a print completes.

This may or may not be related, but when the printer first turns on, it goes to the active print screen showing temperatures for some reason. I don't think it actually turns heating on for anything, but it shows the temperatures, pause, and cancel buttons.

**To Reproduce**
Steps to reproduce the behavior:
1. Which g-codes can cause this issue?
N/A
2. Are you printing from USB or the SD card?
N/A
3. Are you using the LCD display to start prints or are you using commands from the USB?
N/A
4. What OS are you using?
N/A
5. What software are you using?(Cura, Pronterface, Octoprint, Repetier...)
N/A
6. Did this work on the stock Malyan firmware?
Yes
7. What error message is reported?  Are the LED lights blinkly quickly.
N/A
8. Post the full-code and status log, if you can.
Where would I find that?

**Expected behavior**
The Home Axes button should always home the nozzle. The printer should boot into the regular menu, not the active print menu

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
N/A
 - Software: [Cura, Pronterface, Octoprint, Repetier...]
N/A
 - Version [e.g. 22]
33, I believe

**Additional context**

mcheah, 2019-06-14:
Can you confirm that you are running 1.3.3?
There may be a delay between turning on the printer and when the UI first responds.
Also try to reproduce the issue with the SD card removed and see if it changes behavior.  (Specifically in curious if auto.g is on the sd card)

Serneum, 2019-06-14:
I do have your latest version of 1.3.3 for sure. The SD card does have auto.g, I believe, but I will verify and try without the SD card once I get home in a few hours

mcheah, 2019-06-15:
I suspect that's the issue.  It will run auto.g on startup but there's some bug related to it.  You can try adding M73 P100 to the end of it, but tbh is probably just delete the file unless there's some specific reason you need it.

Serneum, 2019-06-15:
That seems to have solved both issues, actually. Thanks!

Serneum, 2019-06-15:
I spoke a little too soon. It looks like the issue of the homing of the axes is only after a print. It showed itself on startup because of the auto.g file, apparently. It still happens after a print is canceled, and I'm not sure if it happens if a print is successful

mcheah, 2019-06-15:
Can you share a video of the exact movement when pressing the home button?  I don't have a printer readily available to replicate.
Is the behavior the same if you issue G28 over USB post print?

Serneum, 2019-06-15:
Sure, here you go. The good news is that I seem to only be able to reproduce after the first print. I know before removing auto.g, it seemed to happen after most prints, but those may all have been "first" prints since I tend to shut off my MPMD after each print since I don't print much or often

[HomeAxes.zip](https://github.com/mcheah/Marlin4MPMD/files/3292677/HomeAxes.zip)

I tried a couple things since you mentioned G28. I tried printing from Octoprint, then sending G28 from the Octoprint terminal and it worked fine. I also tried printing from the SD card directly and then sending G28 from the Octoprint terminal and that worked fine. Trying to print from the SD card and then going to Home Axes seems to be the culprit.

Side note: Sometimes when canceling a print in Octoprint and then canceling it on the printer (since it doesn't seem to do so when Octoprint cancels) results in the screen going a bit nuts. Probably not super important since it's a third party tool, but thought it might be worth mentioning.
[image]

mcheah, 2019-06-15:
Interesting.  I'll check this out next week when I get back.  I suspect this has something to do with the LCD UI driver, which very well could have some state-machine related bug.  A few things to note:

A) the sound I hear in your video sounds more like the extruder than homing the axes.  As an addiitional test, instead of clicking home, can you click the extruder menu and see if it is ignored as well?

B) You have a much older version of the LCD UI (because of the color menu).  There's a chance it could operate in a manner that's different than the newer ones.  I of course can't test your UI version directly.  But here's one more thing you can try: Add `M111 S18` to your start g-code in the file you print initially.  This will send all messages back and forth from the LCD to the print log.  If you do this and then save the entire log from octoprint, that will be really helpful in debugging what's happening

C) The inconsistent UI state for the LCD is a hard one to fix as synchronizing the printer state from Octoprint and the LCD is not trivial.  As a quick and dirty fix, you can add `M117 {SYS:HOME}` to your octoprint cancelling scripts and that should trigger it to go away.  `M524` or `M73 P100` should also work for you in that regard.  Again, debug logs like mentioned above will be the best way to nail down these issues.