mcheah/Marlin4MPMD #32: Putting together a build with serial pins for the only interface.

mmiscool, 2019-06-28:
This is not so much an issue so much as it is a request for help producing a custom build of this firmware. 

I have a situation where there is an EMI problem with the USB interface. In the past on kossel style printers running normal ramps boards there was a way to bypass the USB to serial interface by just connecting to the serial pins. As I understand it the stock firmware has a serial interface that uses the pins that connect to the LCD wifi display board. I am attempting to replace the lcd board with a raspberry pi and a custom python script that can send the gcode over to the machine.

I have not had luck using the 500000 baud 7 configuration for sending gcode and getting the standard acknowledgement. USB interface works fine in an environment with out the EMI concern but I have been unable to get it to work with the serial pins broken out for the display. 

The python script I have been using works with the 3d printer printer over the USB interface and has been previously used with the serial pins on older atmel based ramps boards. 

If it were possible to produce a build with a standard 57600 baud speed for the serial pins and disabling the code to drive the display and USB it would really make my life much easier.

I looked at the wiki here and was unable to understand how to set up the build environment for these arm based devices. 

If there was any chance of preparing a binary that i can flash using the boot-loader mode I would greatly appreciate it. 

-Mike

mcheah, 2019-06-28:
What version of the mainboard do you have?  The original mainboard with the 8x stepper drivers had an unused 5-pin port that included a pair of pins for breaking out a serial port.  This is the most straightforward way to implement what you're asking as it's a #define in config_STM.h.  But the downside of this approach is that it disables the LCD screen (not that it has too, I'm just too lazy to code up two UART drivers), which I assume you're okay with because you're trying to pipe in data through the LCD port.

The LCD port does not accept commands on my firmware, but the stock firmware might.

Another option is to use the wifi chip on the LCD to stream commands via websockets.  @xC0000005 might have some more info on this, but I've not personally done it.  Again, this method would work for stock firmware but not mine.

If you can confirm which mainboard you have (post a picture if you're not sure) I can see what I can do in terms of a custom build.  If you want to try your own hand at compiling, see [here](https://github.com/mcheah/Marlin4MPMD/wiki/Build) for some more details.

A third option for you may be to continue to use the USB port for controlling and sending data, but using my [Fast Uploader](https://github.com/mcheah/MarlinBGC) to send the entire file to the SD card.  Then you won't have to worry about EMI screwing up your prints because the USB port will be idle.

mmiscool, 2019-06-28:
Hello,
Thanks for getting back to me so quickly. 

The way I am using the printers the LCD screens are removed to accommodate an automatic part removal system. I have a raspberry pi that is running a custom gcode sender that manages this and activates large motor that is causing the EMI problem. In the past to get around this I used the serial pins and it worked well on the older kossel style machines using more standard arduino type ramps boards. 

Doing some thing where the gcode is cached to the SD card would not be ideal in this workflow. 

The main board I have looks like this. 
[image]
I have 2 of these machines and know that there must have been a firmware and wiring change at some point as if I swap the control boards between 2 of them the motors go in revers. 

If you do paypal donations I would be willing to send you some cash for your trouble building a couple of binaries to accommodate both of these motor configurations and making the serial pins normally use for the display to be the only method of talking to the printer.

mcheah, 2019-06-29:
Okay.  I understand what you're doing now.  It's pretty straightforward to re use the LCD serial port to do what you need.  PM me on gitter.im, facebook, or reddit and we can discuss the details.

mmiscool, 2019-06-29:
I sent you a msg on gitter.im

closed 2019-06-30 by mmiscool