mcheah/Marlin4MPMD #33: hot end crashes into build plate during G28

JonathanOBrien, 2019-07-02:
When I run the homing cycle after installing the firmware, the hot end moves down until it crashes into the build plate and starts making grinding noises.

After reverting back to the stock firmware, it still does the same thing,  I've tried most every command in the wiki, not sure if I missed a step or if there is a configuration wrong.

Here is the M505:

echo:Steps per unit:
echo:  M92 X114.28 Y114.28 Z114.28 E97.00
echo:Maximum feedrates (mm/s):
echo:  M203 X150.00 Y150.00 Z150.00 E50.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X800 Y800 Z800 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z20.00 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Endstop adjustment (mm):
echo:  M666 X0.00 Y0.00 Z0.00
echo:Delta settings: L=diagonal_rod, R=radius, S=segments_per_second, ABC=diagonal_rod_trim_tower[123]
                DEF=diagonal_rad_trim_tower[123],XYZ=delta_tower_angle_trim[123]
echo:  M665 L120.80 H125.00 R61.70 S200.00 A0.00 B0.00 C0.00 D0.00 E0.00 F0.00 X-120.00 Y-120.00 Z-120.00
echo:PID settings:
echo:  M301 P45.00 I0.07 D1000.00
echo:  M304 P80.00 I12.71 D300.00
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-0.60`

mcheah, 2019-07-02:
I've seen this on at least one other printer.  It seems Malyan is now shipping with steppers in the reverse polarity.  Try [M562 X Y Z](https://www.mpminidelta.com/invert_motor_direction) on stock firmware to swap motor direction.

hitekagpilot, 2019-07-08:
I have/had this exact same problem.  Found this thread and was able to fix it with the stock firmware... but decided I would see if there was a work-around -- and I have it working.

I have a recently new MPMD .. originally with V44 firmware.

Encountered the same issue as 'JonathanOBrien' above after installing the firmware (V43+).

Flashed to the Stock V45 firmware.. still crashing.  Found this thread and from the Terminal in Octoprint was able to perform the M562 XYZE followed by an M500 and a reboot (power-off).

Printer was back to operating....

Then decided to do flash back Stock Firmware, flip the unit upside down, pull the bottom panel off.
One at a time (so as not to confuse which servo wire-loom is which....
with a pair of needle-nosed pliers, I pulled and 180'd the plastic connectors from the board, slid them back onto the pins, then plugged the cables back in.  All four (4) axis (X, Y, Z and E).

From the stock V45 firmware, performed an M502 -- reload settings from firmware (Factory reset Firmware).... performed an M503 just to check if the stepper reversal was still active and it was not, NEXT (and with my hand of the power connector) did a G28... and IT WORKED!!! Sent a G29 and the ABL processed.

Next, I put the Marlin4MPMD v43+ 10A fastSPI firmware and flashed that.  I had a spare 12v 30A power supply laying around, so weeks back I had wired that to a barrel connector for the MPMD for power (a little overkill).  Again (with my hand on the power connector) I sent a G28 followed by a G29 with success.

Now I will be sending a test cube and a bigger print job.

HiTekAgPilot

Modded Monoprice Mini Delta
- All-metal V6 hotend (Gulfcoast robotics)
- MP Mini Delta E3D V6 Hotend Replacement (mount) by evancli
- Stock 30mm Hotend cooling fan
- 5015 12v dual-ball bearing blower fan (Amazon)
- 120mm x 3mm borosilicate glass plate (Amazon)

mcheah, 2019-07-08:
Glad this works, can you find some of the markings on the side of the stepper drivers so that i can confirm the model?
I'll work on a fix for reversing on software.

JonathanOBrien, 2019-07-09:
That fixed it for me as well.  Thanks!

mcheah, 2019-07-20:
It looks like you can correct this by manually entering
M92 X-114.28 Y-114.28 Z-114.28 E-114.28

PurpleHullPeas, 2019-09-22:
Just thought I'd drop by to update users who are still using this thread to debug this issue.

First of all, @mcheah had this to say in another thread as of 1.3.3.

"The negative m92 fix doesn't work because it flips which endstops trigger.  Right now the only way to fix with marlin is to physically reverse your wiring.  It's on my todo list but i honestly don't have a whole lot of time for development these days."

I recorded a short, low-quality, video on solving the issue, both stock firmware and hardware.

https://youtu.be/RQnMoen0l6A

FLamparski, 2019-10-02:
I can confirm that swapping the XYZ motor wires around fixes the issue. No word on the extruder yet but will update when I manage to get filament coming out of the nozzle again.

PurpleHullPeas, 2019-10-02:
> 
> 
> I can confirm that swapping the XYZ motor wires around fixes the issue. No word on the extruder yet but will update when I manage to get filament coming out of the nozzle again.

After making that video, I learned that some people need to swap XYZE, while others only need to swap XYZ. The good thing about the method shown in the video, is that it's very easy to unswap the connectors on the board.

FLamparski, 2019-10-02:
Yep, mine's XYZE. I think the telltale sign is in running `M503` on the stock, unmodified firmware:

```
>>> M503
SENDING:M503
...snip...
echo:Invert axis: M562 XYZE
XYZABCD+++----
...snip...
ok N0 P15 B15
```

I guess that's from the M562 performed at the factory.

twagler, 2019-10-12:
I only needed to swap XYZ. Extruded was fine.