mcheah/Marlin4MPMD #34: Tower arrangement graphic is incorrect

gigawatts, 2019-07-21:
**Describe the bug**
I think the [calibration tower arrangement](https://github.com/mcheah/Marlin4MPMD/wiki/Calibration#m665m666-delta-parameter-calibrations) graphic is incorrect. In the [migration instructions](https://github.com/mcheah/Marlin4MPMD/wiki/Migration#migrating-from-stock-firmware) you state "By default the towers are rotated 120deg to the right to correctly map +Y opposite the LCD screen and and +X to the right of the LCD screen." (BTW you have a duplicate 'and' in that sentence)

The ```G1 X0 Y50``` command given in the calibration instructions moves the extruder towards the tower opposite the LCD, but your ASCII art depiction of the towers and their coordinates does not agree with your statement. The diagram infers that ```G1 X0 Y50``` command should move the extruder towards the tower to the right side of the LCD (as viewed from the outside of the printer, viewing the LCD).

mcheah, 2019-07-21:
Yes this is correct, I haven't bothered to update the documentation.  The wiki is open, feel free to update yourself

gigawatts, 2019-07-30:
Are your tower labels (T1, T2, T3) in your graphic correct? What determines the tower number, or are they just arbitrary labels you gave them for descriptive purposes?

mcheah, 2019-07-30:
Use 
```
M666 X-20 Y0 Z0
M666 X0 Y-20 Z0
M666 X0 Y0 Z-20
```
as a guide and it'll become clear which one is tower 1 2 3
tower numbers are just an arbitrary way to bind all M666/M665 settings together