mcheah/Marlin4MPMD #35: Surpasses Endstops during G28

ApexProxy, 2019-07-26:
**Describe the bug**
Arms surpass end stops when G28 is issued

**To Reproduce**
Steps to reproduce the behavior:
To start off, I just upgraded from stock V44. I previously ran M92 X114.28 Y114.28 Z114.28 E97 with success. After flashing Marlin4MPMD I issued M502 and reset everything. At first issuing G28 caused my hot end to crash into my build plate so I followed the suggestions of another issue reported. I started by setting the negated M92 settings one by one to see what the movement did and each stepper reversed as expected. However now when I issue G28 instead of crashing into the build plate it crashes into the top, completely ignoring the end stops. I also noticed that everything seems to completely seize up when I use the full M92 X-114.28 Y-114.28 Z-114.28 E-97 vs omitting the extruder reversal (E-97). I have no idea why that setting would break the axis steppers completely and I am pretty lost at this point.

Non-stock hardware:
10A PSU
Borosilicate+PEI Build Plate
OctoPi via Zero W

On a side note I did a PID calibration on both the hot end and bed and WOW what a difference in speeds, this is exactly the reason I sought Marlin4MPMD in the first place haha.

mcheah, 2019-07-26:
When you manually position the carriage at the top and issue M119, do the endstops trigger?

mcheah, 2019-07-26:
Tbh the easiest solution is to remove the wiring from the connectors and reverse them.  Others have taken off the plastic connector on the board and flipped the connector on the bare pins

ApexProxy, 2019-07-26:
I'm not as concerned about the revered steppers as I am the lack of endstop function but I will probably do something like that once I get the endstop issue resolved. I will try M119 when I get home in a couple hours and post the results. Thanks for responding BTW!

ApexProxy, 2019-07-27:
Alright so M119 shows X,Y,and Z max endstops triggered with a raised carriage. For the sake of curiosity I ran it again with the carriage lowered and my finger on the build plate sensor and it showed all open other than Z min so its reading them all correctly.

ApexProxy, 2019-07-27:
Apparently negating the M92 step settings was not working for my printer at all. I flipped all the stepper plugs and it all works great now. I was hesitant to do so at first just in case the issue was elsewhere but ended up just giving it a go and sure enough haha. It definitely seemed like the printer thought it was moving the opposite direction that it really was so that makes perfect sense. Thanks again.

mcheah, 2019-07-27:
I'll investigate this further and try to come up with a solution.  It'd be better if monoprice didn't keep changing things up every 6 months to make my life harder.

ApexProxy, 2019-07-28:
Yeah, that's kind of the problem with cheap Chinese electronics. They have a tendency of changing things for no apparent reason and no real clear way of telling what was actually changed. I think part of this is due to frequent changes in supply so they end up just making quick fixes just to get things running even if it's very inefficient.

closed 2019-07-28 by ApexProxy