mcheah/Marlin4MPMD #36: Octoprint connection first attempt failure

ApexProxy, 2019-07-28:
**Describe the bug**
On my first attempt to connect OctoPrint I get an error and the connection times out. The second attempt succeeds however. The issue appears to happen with or without the Malyan/Monoprice Connection Fix Plugin. Before Marlin4MPMD I was able to connect on the first try consistently using the plugin fix. I also noticed that it doesn't auto connect on server startup like it did before (at least I am fairly certain it used to lol). However nothing is displayed in the terminal regarding that.

**To Reproduce**
Steps to reproduce the behavior:
1. Which g-codes can cause this issue? NA
2. Are you printing from USB or the SD card? NA
3. Are you using the LCD display to start prints or are you using commands from the USB? NA
4. What OS are you using? OctoPi 0.16.0 
5. What software are you using? OctoPrint 1.3.11
6. Did this work on the stock Malyan firmware? Yes, v44
7. What error message is reported?  NA
8. Post the full-code and status log, if you can. 
This is both attempts, failure and then success:

Changing monitoring state from "Offline" to "Detecting serial port"Serial port list: ['/dev/ttyACM0']Connecting to: /dev/ttyACM0Using Malyan/Monoprice Connection Fix Plugin to create serial connectionChanging monitoring state from "Detecting serial port" to "Opening serial port"Connected to: Serial<id=0xb0762a30, open=True>(port='/dev/ttyACM0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitorChanging monitoring state from "Opening serial port" to "Connecting"Send: N0 M110 N0*125Recv: Error:No Line Number with checksum, Last Line: 0There was a timeout while trying to connect to the printerChanging monitoring state from "Connecting" to "Offline"Connection closed, closing down monitorConnecting to: /dev/ttyACM0Using Malyan/Monoprice Connection Fix Plugin to create serial connectionChanging monitoring state from "Offline" to "Opening serial port"Connected to: Serial<id=0xafc15850, open=True>(port='/dev/ttyACM0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitorChanging monitoring state from "Opening serial port" to "Connecting"Send: N0 M110 N0*125Recv: okChanging monitoring state from "Connecting" to "Operational"Send: N0 M110 N0*125Recv: okSend: N1 M115*39Recv: FIRMWARE_NAME:Marlin4MPMD 1.3.3-MPMD (Github) SOURCE_CODE_URL:https://github.com/mcheah/Marlin4MPMD PROTOCOL_VERSION:1.0 MACHINE_TYPE:MPMD EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ffRecv: ok

**Expected behavior**
Pressing Connect should just connect the first try.

**Screenshots**
NA

**Desktop (please complete the following information):**
NA

**Additional context**
I can repeat and post the results without the connection fix plugin if needed. 

Side note; Even with these occasional hiccups I am extremely grateful for this firmware. I could never hold temps nearly this perfectly before and gotta love the new G29 lol

mcheah, 2019-07-28:
You can try disabling "Commands that always require a checksum" in the octoprint options.  You shouldn't need the malyan connection fix with this firmware, I don't think that's your issue but I thought I'd point it out.
I haven't tested with anything newer than octoprint 1.3.9, so it's entirely possible there's an incompatibility issue there.  I also haven't tried out the auto-connect option.

Is this an issue when the printer first powers up?  Or will this happen every time you disconnect/connect?

ApexProxy, 2019-07-28:
Ok, so disabling the malyan fix didn't help but I'll leave that disabled. The issue was only present when the printer first powers up. After that each connect/disconnect cycle was one-shot. I have removed M110 from the checksum requirement list and that seems to have done the trick but I'll let you know if it comes back on another cycle. In addition, auto-connect is now functioning again. Thanks for helping.

ApexProxy, 2019-07-28:
Update: The exact same error happens when I restart the printer and leave Octoprint up. It takes two tries to connect.

mcheah, 2019-07-29:
looks like there's some gibberish on startup when you first power on the printer.  Honestly, the best solution might be a custom octoprint plugin that sends a junk command before connecting.  I think there's two things happening:
The printer doesn't initialize the USB connection right away, so octoprint is probably sending data before it's ready, and in addition it boots up with some garbage commands in the buffer on startup so the first response to octoprint is probably not what it's expecting.  If you find out more, let me know, but I'll be honest this is a pretty low priority fix at the moment.

ApexProxy, 2019-07-29:
So I tried waiting a bit after restarting the printer and it is the same so I'm not too certain what it is. I think my best fix for it is to avoid restarting the printer unnecessarily. TBH I mostly see this error now when I derp something up and have to power cycle the printer haha. Overall I am sure you have bigger fish to fry so I get not really putting much priority to this.