mcheah/Marlin4MPMD #45: M600 Filament change

realbrianosaurus, 2019-08-10:
I've been doing multi-color prints by changing filament at a certain layer. I've been using the M600 command on the stock firmware (I have one printer on stock v44, and one on latest Marlin4MPMD). M600 can specify an X,Y to park the print head, and a Z offset to raise above. You can also specify how much to retract before moving, how much to retract to Unload the filament (have not tested that in the stock firmware), and how much to extrude to Load. When you press "resume", it drops back down by the specified Z offset, and returns to where it left off on the print.

Stock firmware has a decent "M600" implementation for the stock build plate, but it seems broken when the build plate is not stock height (like adding a glass plate on top). I send it to X55, Y0, Z20. Then I pull the old filament and push the new one in by hand until the new color starts coming out. Then manually retract to relieve pressure. I set U0 and L0, since I'm doing that part manually. Works great for changing colors.

With a glass plate on the bed, pausing stock firmware seems to plunge the extruder down about the depth of the plate before it moves up to pause. When it resumes the print, it starts about the glass thickness above the last printed layer. It seems to know where Z==zero "should" be, and pause works off that, instead of the actual location of the surface set in M666 or the height that it probed in G29. Pausing (of all varieties) seems to be implemented poorly in the stock firmware, and I was hoping Marlin4MPMD did it better (like everything else it does :).

aegean-odyssey, 2019-12-20:
As far as I can tell, the Mini Delta's LCD display seems to keep its own "protected" states -- pause/resume, running, waiting, etc. -- making it difficult (maybe impossible) to use the display  to interact with the user in any generic fashion.

I've ported the latest Marlin firmware (bugfix-1.1.x / 1.1.9) to Monoprice Mini Delta _(see [mpmd_marlin_1.1.x](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x))_, and have run into similar problems to ones that you describe.

The port does handle pause (with park), resume, cancel (with home) when the user initiates these actions via the LCD display, but otherwise the display is fairly useless. With the display's limitations, commands such as M0/M1 (wait for user input), M600 (change filament), that is, commands that require user intervention just do not work well with the display. (IMHO)

As a work-around, I've created a single LED/pushbutton UI -- the LED flashes when expecting/waiting for the user to provide input (that is, press the button). So with this simple UI, the commands M0/M1 and M600 can be realized. (In keeping with Marlin, M108 command will also serve as a button press.)

I think this is a workable solution. And though very little testing has been done with the M600 command, M0/M1, pause, resume, cancel seem to work fairly reliably. If interested, I encourage you to look at the [mpmd_marlin_1.1.x](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x) repository. I've been printing with the firmware for some time, but please keep in mind, that it is a "work-in-progress"

mcheah, 2019-12-20:
Glad to see someone else working on this.  You may be better served starting from 2.0 because they have native support for HAL and should adjust have stm32F1 support.

The issue with the LCD is that it's running it's own closed source firmware which nobody has made a replacement for yet.  So to add simple functionality requires redoing the entire UI.

aegean-odyssey, 2019-12-21:
Agreed on the LCD. With a little more technical info on the Malyan LCD, perhaps someday I'll tackle the project. For now, though, I'm working on getting the most out of display. I'm almost there -- status and controls work, directory listings fully implemented, limited messages. I need to learn how to set the WIFI SSID, WIFI password, etc -- what the LCD needs to implement the functionality of the M550, M551, M552, M555, M556, M563 commands. 
Any ideas?  

And will likely migrate to Marlin2, but I wanted to start the port from known working source code. Turns out, Marlin requires very little from HAL (2 timers, no dma, SPI, etc) so a "board level interface" (BSP) was pretty easy to implement directly.

mcheah, 2019-12-21:
You can get most of those by spying in the UART traffic with the stock firmware, i should warn you that if your goal is faster more reliable uploads from wifi, that's a dead end.  Hit me up on gitter or facebook messenger and we can discuss more details.