mcheah/Marlin4MPMD #47: Heat failure errors being thrown.

michaelhill, 2019-10-12:
**Describe the bug**
Generally, heating to PLA temperatures (195 deg) doesn't cause issues, but I cannot get to ABS temperatures (230) without a heating failed error. This sometimes occurs when trying to get to PLA temperatures, but is less common. I had this issue when using the default PID gains baked into the firmware as well as both PID settings suggested by Dennis Brown here: https://www.facebook.com/groups/mpminideltaowners/permalink/2042879919060817/?hc_location=ufi

> HERE ARE MY VALUES:
Fastest settling with some overshoot:
M301 P45 I0.07 D1000
Slower settling with less overshoot:
M301 P25 I0.05 D1000

Readout of the logs are as follows:

> Send: M104 S230
Recv: ok
Send: M105
Recv: ok T:19.0 /230.0 (3945) B:19.1 /0.0 (3944) @:127 B@:0
Send: M105
Recv: ok T:19.6 /230.0 (3941) B:19.1 /0.0 (3944) @:127 B@:0
Send: M105
Recv: ok T:21.3 /230.0 (3929) B:19.1 /0.0 (3944) @:127 B@:0
Send: M105
Recv: ok T:23.9 /230.0 (3910) B:19.1 /0.0 (3944) @:127 B@:0
Send: M105
Recv: ok T:27.3 /230.0 (3882) B:19.1 /0.0 (3944) @:127 B@:0
Send: M105
... (Shortened for brevity)
Recv: ok T:222.7 /230.0 (288) B:19.9 /0.0 (3939) @:51 B@:0
Send: M105
Recv: ok T:223.7 /230.0 (283) B:19.9 /0.0 (3939) @:0 B@:0
Send: M105
Recv: ok T:224.7 /230.0 (279) B:19.8 /0.0 (3939) @:0 B@:0
Send: M105
Recv: ok T:225.1 /230.0 (277) B:19.8 /0.0 (3939) @:0 B@:0
Send: M105
Recv: ok T:224.9 /230.0 (278) B:19.8 /0.0 (3939) @:112 B@:0
Send: M105
Recv: ok T:224.3 /230.0 (280) B:19.9 /0.0 (3939) @:127 B@:0
Recv: Error:Heating failed, system stopped! Heater_ID: 0
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Operational" to "Error: Heating failed, system stopped! Heater_ID: 0 - Printer halted. kill() called!"
Changing monitoring state from "Error: Heating failed, system stopped! Heater_ID: 0 - Printer halted. kill() called!" to "Offline (Error: Heating failed, system stopped! Heater_ID: 0 - Printer halted. kill() called!)"
Connection closed, closing down monitor

**To Reproduce**
Sending M104 S230 to set the hot end temperature to 230 degrees seems to be all I need to do

**Expected behavior**
The hot end should heat to 230 degrees without error

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Microsoft Windows
 - Software: Cura -> OctoPrint (running on BeagleBone Black)
 - Printer: MonoPrice Mini Delta (v44 stock firmware)
 - Version 1.3.3 v43+, 10A
 - Power supply: HJDZ-1210 120W 12V Adapter (https://www.amazon.com/gp/product/B07FFKL7CR/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1) 

**Additional context**
I notice in the screenshot there are oscillations as the temperature approaches the setpoint (within the 20 degree PID range). This makes me wonder if this section of code is what is throwing the error:

```
temperature.cpp:

    // Check if the temperature is failing to increase
    #if ENABLED(THERMAL_PROTECTION_BED) && WATCH_BED_TEMP_PERIOD > 0

      // Is it time to check the bed?
      if (watch_bed_next_ms && ELAPSED(ms, watch_bed_next_ms)) {
        // Has it failed to increase enough?
        if (degBed() < watch_target_bed_temp) {
          // Stop!
          _temp_error(-1, PSTR(MSG_T_HEATING_FAILED), PSTR(MSG_HEATING_FAILED_LCD));
        }
        else {
          // Start again if the target is still far off
          start_watching_bed();
        }
      }
```

The heaters may need more physical characterization to better tune PID values out of the box. Is there a variety of heater cores used across the MPMD line? Alternatively, there are somewhat more advanced (but not too complicated) implementations of PID that should be considered (such as selective integral ranges to prevent windup). However, this is probably a separate issue.

mcheah, 2019-10-12:
Try resetting to default PID values and see if that helps.  In general you shouldn't trigger thermal runaway on initial heating, so that's somewhat confusing.
Also try heating to 200C and see if that shows same behavior.  Please continue to post logs and temperature charts

michaelhill, 2019-10-12:
[image]

OK, strange. manually setting the default PID value with "M301 P20 I1.60 D56.77" seems to have fixed my issue. I was originally having issues with the hot end to get to temperature, which is why I tried out Dennis's values, but this seems to not be an issue. I'll close and create a new issue if I see any further problems.

closed 2019-10-12 by michaelhill

mcheah, 2019-10-12:
Okay, so you're right, the above code is what's triggering the temp error, not thermal runaway protection.  The two operate in similar principles, but the gist of what's happening is that Marlin expects during initial heating your hotend should rise at least 2 degrees every 20 seconds, otherwise it assumes your thermistor has fallen out and something has gone wrong.  Your PID settings are very aggressively tuned to underdamp the temperature signal, which then causes the temperature to increase at a very low rate, which ends up triggering the thermal protection.  Aim for a little more overshoot and you should avoid this error

mcheah, 2019-10-12:
FYI, the PID values are _not_ comparable between stock firmware and Marlin, as the temperature timers are completely different.  If you had copied somewhere else as a starting point it may have screwed up the whole tuning process.  Glad to see you got it working though.