mcheah/Marlin4MPMD #48: M500 command does not save my setting to EEPROM (Marlin4MPMD)

holt-ace, 2019-10-23:
**Describe the bug**
It’s the oddest thing. I plug my computer in and go through all the commands and mesh bed leveling (Marlin) and after everything works fine I type my M500 command and nothing happens.

I’ve let my computer sit for 5,10,15 minutes “sending M500 command” and nothing. So every time I print it re-runs the 7x7 bed leveling calibration.

**To Reproduce**
Steps to reproduce the behavior:
1. Which g-codes can cause this issue? 
N/A
2. Are you printing from USB or the SD card? 
SD card/USB rarely
3. Are you using the LCD display to start prints or are you using commands from the USB?
LCD prefferably
4. What OS are you using?
MAC OS Catalina 10.15
5. What software are you using?
Print-Run (apparently i cannot use this anymore with my new OS)
6. Did this work on the stock Malyan firmware?
Yes i could save via M500 just fine.
7. What error message is reported?  Are the LED lights blinkly quickly
NO error report
8. Post the full-code and status log, if you can.

**Expected behavior**
I expect my mesh to be saved to the EEPROM and not have to re-do the 7x7 leveling everytime i print.

**Desktop (please complete the following information):**
 - OS: MAC OS Catalina
 - Software: [Pronterface (need to change this)]
 - Version 10.15

**Additional context**
Add any other context about the problem here.

mcheah, 2019-10-23:
Can you provide the full pronterface logs (enable debug communications)?

holt-ace, 2019-10-23:
I was trying to find them. I updated my OS yesterday and pronterface is not supported on the new OS it says "it needs an update". So I have no access to my previous logs.

Can I try to do the same thing in Cura?

mcheah, 2019-10-23:
No.  You can try repetier though

mcheah, 2019-10-26:
I'm going to close this since the issue was that you had G29 P2 in the starting gcode.  If you continue to have issues restoring from EEPROM, please re-open the issue

closed 2019-10-26 by mcheah