mcheah/Marlin4MPMD #52: Raise hotend when pause or cancel is triggered from the user interface display

ramesh66666, 2019-12-09:
when pause and home are pressed from the user interface the nozzle stays where it is and damages the model being printed. I do not have October print and I'm printing from an SD card are there any solutions to raise the hotend when when a pause is triggered from the user interface display.

Is it possible to to make the hotend move to a particular point at the top of the machine when a pause is triggered. It is not not so important when when the print is cancelled but it is a real pain if if you want to pause the print to do a filament change for example as the nozzle will not move during the pause.

mcheah, 2019-12-09:
This is a duplicate of #25.  While it seems trivial to implement it gets complicated keeping track of the system state under all conditions and can result in the head crashing if you're not careful.  The best solution is octoprint, which should have an option to set g-code commands on-pause as long as you setup //actions.

closed 2019-12-09 by mcheah

ramesh66666, 2019-12-10:
Okay thanks for your help and time