mcheah/Marlin4MPMD #54: G29 P0 - remove asking for homing before

michtill, 2020-01-10:
G29 P0 starts with homin it self. But it needs to call G28 before. I think, it is not neccerarry. 
In start GCode the G29 P0 is enought. Isn't it?

mcheah, 2020-01-10:
I'm not 100% what your question is, but in a nutshell homing needs to be built into G29.  I can get into the details, but keeping track of all the bed leveling state is complicated and the easiest solution.

ppetermann, 2020-02-23:
I believe what @michtill means is that there is a check for G28 to have run, which might be a bit pointless if G29 does the homing anyways because this way we end up always having the homing done twice.

mcheah, 2020-02-23:
G29 P1 doesn't require it but i could certainly change the logic