mcheah/Marlin4MPMD #55: M500 not working

aaMarcinaa, 2020-01-27:
Hello.

After migration from stock to Marlin firmware, and after preforming G33 I was try save new value to eprom.
Unfortunately, not working.
Also printer try print 2.6mm above bed ;-)

In detail:
SENDING:G33 P1
Bed X:0.00000 Y:0.00000 Z:-2.23766
Bed X:0.00000 Y:50.00000 Z:-1.78216
Bed X:-43.30054 Y:-24.99958 Z:-2.09717
Bed X:43.30054 Y:-24.99958 Z:-1.54589
Start M666 X0.00 Y0.00 Z0.00
End M666 X-0.28 Y-0.65 Z0.00

Next I ty  M500 and nothing happening.

>>> M500
SENDING:M500

Unfortunately, not working. - is hang on this state for minutes.
This same PID for printhead not saving etc.

Card in socket, connected by  usb, pronterface as host.

mcheah, 2020-01-27:
Are you saying it never gets an "OK" response from the printer and hangs?
Do any of the leds come on when this happens?
Which version of the firmware was originally on your printer?  Approximately when did you buy it?
Can you send M111 S127 repeat the test including full log for M503/M500/M503?

aaMarcinaa, 2020-01-27:
Printer is month old, v44

Connecting...
Printer is now online.
>>> M500
SENDING:M500
>>>  M111 S127
SENDING:M111 S127
echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION
echo:M105
echo:M105
echo:M105
echo:M105
>>> M500
SENDING:M500
echo:M500
echo:M105
echo:M105
echo:M105
echo:M105
>>> M503
SENDING:M503
echo:M503
echo:Steps per unit:
echo:  M92 X114.28 Y114.28 Z114.28 E97.00
echo:Maximum feedrates (mm/s):
echo:  M203 X150.00 Y150.00 Z150.00 E50.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X800 Y800 Z800 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate
(mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z
jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z20.00 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Endstop adjustment (mm):
echo:  M666 X0.00 Y0.00 Z0.00
echo:Delta settings: L=diagonal_rod, R=radius, S=segments_per_second,
ABC=diagonal_rod_trim_tower[123]

DEF=diagonal_rad_trim_tower[123],XYZ=delta_tower_angle_trim[123]
echo:  M665 L120.80 H122.30 R61.70 S200.00 A0.00 B0.00 C0.00 D0.00 E0.00
F0.00 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P45.00 I0.07 D1000.00
echo:  M304 P80.00 I12.71 D300.00
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-2.60
echo:M105
>>> M500
SENDING:M500
echo:M500
echo:M105
>>> M503
SENDING:M503
echo:M503
echo:Steps per unit:
echo:  M92 X114.28 Y114.28 Z114.28 E97.00
echo:Maximum feedrates (mm/s):
echo:  M203 X150.00 Y150.00 Z150.00 E50.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X800 Y800 Z800 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate
(mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z
jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z20.00 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Endstop adjustment (mm):
echo:  M666 X0.00 Y0.00 Z0.00
echo:Delta settings: L=diagonal_rod, R=radius, S=segments_per_second,
ABC=diagonal_rod_trim_tower[123]

DEF=diagonal_rad_trim_tower[123],XYZ=delta_tower_angle_trim[123]
echo:  M665 L120.80 H122.30 R61.70 S200.00 A0.00 B0.00 C0.00 D0.00 E0.00
F0.00 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P45.00 I0.07 D1000.00
echo:  M304 P80.00 I12.71 D300.00
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-2.60
echo:M105
echo:M105
echo:M105

aaMarcinaa, 2020-01-27:
And after return to default firmware:
>>> m501
SENDING:M501
echo:V24 stored settings retrieved (244 bytes)
echo:Steps per unit:
echo:  M92 X56.50 Y56.50 Z56.50 E48.50
echo:Maximum feedrates (mm/s):
echo:  M203 X150.00 Y150.00 Z150.00 E50.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X800 Y800 Z800 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables:
S=Min feedrate (mm/s),
T=Min travel feedrate (mm/s),
B=minimum segment time (ms),
X=maximum XY jerk (mm/s),
Z=maximum Z jerk (mm/s),
E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z20.00 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:Invert axis: M562 XYZE
XYZABCD---+-+-
echo:Endstop adjustment (mm):
echo:  M666 X0.00 Y0.00 Z0.00
echo:Delta settings: L=delta_diagonal_rod, R=delta_radius, S=delta_segments_per_second
echo:  M665 L120.80 R61.70 S500.00
echo:PID settings:
echo:  M301 P20.00 I0.02 D250.00 C100.00 L20
echo:  M304 P10.00 I0.02 D305.40
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z0.00
ok N0 P15 B15
>>> m500
SENDING:M500
echo:Settings Stored (248 bytes)
ok N0 P15 B15

Of course is problem with microstepping, G28 move printhead up to middle of height, but M500 working ;-|)

arcanux76, 2020-01-27:
I hear your pain.  I'm not very knowledgeable on this stuff and I'm having a similar issue.  Something went wrong during the flash from v.44 to Marlin and now I cant reflash or save anything to the eeprom.  When I run M500, it just hangs there SENDING and nothing happens.  Same with M665 S120, it just says SENDING and never completes.  Also, the printer wont reflash cuz it doesnt recognize the fcupdate.flg or the firmware.bin.  I think the bootloader is dead, and the printer will only read the SD card if it was installed prior to power cycling.  If you swap while it powered it wont reread the card.

aaMarcinaa, 2020-01-27:
My situation is not bas as yours.

After boot working with and without SD, im able to change firmware to any
version.

I think, now Im only need change steps and will be ok.
Soon.

mcheah, 2020-01-28:
@aaMarcinaa,
I suspect they may have changed something in recent versions of printer, which means I won't be able to reproduce on my end.  Can you answer some questions?
Does the printer restart itself, as in the LCD screen visibly reboots?
Do any LED's come on when it locks up?
If you unplug/replug the USB cable to the PC and reconnect in Pronterface does the printer start responding?
Do you think you can take the bottom cover off of your printer and take a high resolution picture of your mainboard so I can compare the hardware revision?

mcheah, 2020-01-28:
Also,
Before taking your printer apart can you do the following?
turn off temperature probing in pronterface by sending "monitor off"
Then enable Settings->Debug communications in pronterface and issue the same test as you did initially.
[image]

arcanux76, 2020-01-28:
Well, good news.  I've discovered that it is now saving the data to the eeprom, it's just that its not showing any acknowledgement when it does.  Proved it with some trail and error running M421 and M503.  Only thing that's works flakey is the sd bootloader.  It wont read fcupdates and I have to power cycle the printer for it to read an SD card.  I wonder if the damage was there prior to the flash and just not noticeable.

mcheah, 2020-01-28:
Try issuing M21 instead of power cycling.

mcheah, 2020-01-30:
@aaMarcinaa,
were you able to get the addtional logs?
Looking at your initial results I actually see that your M851 setting is changed from the default, implying that the EEPROM is saving correctly (that it is saved as -2.60 is likely why it's printing high).  I suspect that it looks like it's not working because pronterface scrubs the "ok" responses from the log.
I'm going to go ahead and close this issue, if you continue to have problems feel free to re-open.

closed 2020-01-30 by mcheah