mcheah/Marlin4MPMD #56: It looks like MP released the source code

clytle374, 2020-02-04:
It's not a bug or a feature request.  Just wanted to let people know that supposedly MP released the source code for many of their printers.  Could be handy to developers here. 

https://github.com/Monoprice3DP/MiniV2DeltaSourceCode

Cory

mcheah, 2020-02-04:
Thanks this is indeed very interesting.  However i very much doubt this is the source code for the stock firmware but rather a port from  marlin 2.0.  Interesting nonetheless.

clytle374, 2020-02-05:
That is definitely possible. Just wanted to let people know it was out there in case someone could use it as a reference.

closed 2020-02-05 by clytle374

mcheah, 2020-02-05:
For example they include malyanlcd.cpp which was only added to Marlin once we hacked and reverse engineered the v2 and the mini delta