mcheah/Marlin4MPMD #59: Losing USB connection during Heat-Up

ppetermann, 2020-02-23:
I recently installed Marlin4MPMD (5A version), and ever since computer will lose USB connection to the printer while it is heating up to print - this usually happens between 192C and 197C on the Hotend (with the bed at around 30C at that time), when aiming for 215/40 in cura.

**To Reproduce**
Steps to reproduce the behavior:
1.Open Cura 4.5 (on Win10)
2. Load any STL 
3. Set print settings so the hotend will go to 215, the heatbed to 40 C
4. Start Print via USB

**Expected behavior**
I expect the printer to heat up and printing, as it did on stock

**Desktop (please complete the following information):**
 - OS: Win10, latest patches on 2020-02-23
 - Software: Cura
 - Version:  4.5

**Additional context**
the first thing I notice about the problem is windows playing the sound for a lost connection, cura will actually freeze all temperatures, but not realize the connection is dead, until I pull the USB cable out and put it in again. The printer wont continue, and will lose heat again.
The problem doesn't occure twice (except if i let cool down the printer), and doesn't happen at all, if i preheat the hotend before starting the print.

mcheah, 2020-02-23:
Are you preheating from the LCD, usb, or gcode?
Have you modified pid settings?
Can you try the same procedure with pronterface or octoprint and provide the full serial logs (enable debug communication on pronterface)?
When this happens does the lcd continue to update?  
Does replugging the USB fix the issue?
Also provide a picture of the heating chart?
9/10 it is a misconfigured PID setting, so you can try M502 to reset to default and repeat.
Also cura is notorious for being greedy with the serial port and dropping connections, so i strongly recommend disabling usb printing in the cura addins and streaming gcode with pronterface, octoprint, or repetier-host.

ppetermann, 2020-02-24:
@mcheah usually I don't preheat, but when testing I found that when preheated no issue exists. This is true for both, preheating via the buttons on the printer as well as preheating through cura, which I'd assume is sending gcode via the usb serial connection.

I've tried with the PID settings it had after installing Marlin4MPMD as well as with auto tuning them, the problem exists in both cases.

I don't use octoprint, and I've used pronterface only as a terminal so far. I'm not keen on setting those two up for printing, I've spent more time then I wanted on learning how to use Cura.

Replugging the USB, as described in my original post, the print won't be continued and the printer cools down again, If I immediately start the print again once the connection is back, it will run through without further issues though. The LCD of the printer returns to the main screen when this happens.

I have no idea where I'd find such a heating chart. 

I'm not sure what you mean by "greedy with the serial port", and I've not had any issues with it on stock firmware, which I've been using since December on an almost daily basis. 

The Idea of using yet another software to stream the gcode seems to just complicate things, I don't want to juggle files between multiple applications and all that stuff, life's too short.

mcheah, 2020-02-24:
So typically issues with the usb are indicative of a heater fault or system reset, which is why i like to ask to ask for serial logs and a temperature chart.  At this point i have no way of knowing if it's an issue with cura or a problem with your start gcode or something else in your system, hence why i recommend trying pronterface or octoprint.  You can also try printing with the sd card and see if that also shows the problem.

ppetermann, 2020-02-27:
When printing through the SD Card:
- it did lose the USB connection at a slightly higher temperature (~205) 
- it didn't care and just continued with the print

mcheah, 2020-02-27:
Try preheating manually, remove M109 and M190 from your start gcode, add M73 P0 to start gcode and repeat experiment printing from SD card.
My theory is that it's a cura problem because they don't respect the busy protocol.  So when you start M109, it spams M105 commands before receiving ok response.  This overwhelms the buffer and drops the usb connection.

ppetermann, 2020-02-29:
I'm not sure how you get from "i've run this from sd card" to "cura is spaming m105" cura hasn't even been running.

mcheah, 2020-02-29:
I assumed you were watching the temps with cura while the sd print was going.  How do you know the USB connection is dropping?

ppetermann, 2020-02-29:
> * it did lose the USB connection at a slightly higher temperature (~205) **(as confirmed by audible alert of unplugging a device on MS Windows)**

mcheah, 2020-02-29:
No need to get snippy, I'm trying to help you.
What shows up in device manager when the device disconnects?
If you start a print with the usb unplugged, let it pre-heat and then re-attach the USB later on in the print does it still disconnect?
Can you repeat the process with a different PC?

ppetermann, 2020-03-01:
> What shows up in device manager when the device disconnects?
the entries for the mpmd disappear

> If you start a print with the usb unplugged, let it pre-heat and then re-attach the USB later on in the print does it still disconnect?
- if it starts printing it doesn't disconnect, meaning if i connect the cable after the print has started, it won't disconnect. 
- if i start the print and connect the usb cable while it is heating up, it will disconnect

> Can you repeat the process with a different PC?
no

I've just noticed that the light on the button on the right side of the mpmd flashes white when it disconnects for a moment.

mcheah, 2020-03-01:
Unfortunately I  don't have a great way to help debug this as i have no way to reproduce your issue.  I suspect that your printer may have a bad usb port.  If you resolve the issue feel free to report what you did to fix it.

closed 2020-03-01 by mcheah

ppetermann, 2020-03-06:
one more information: If I don't heat up the bed, but keep it at room temperature, the issue doesn't happen.

ppetermann, 2020-03-20:
I upgraded to a 10A power supply, and the problem went fully away (but I'm 100% certain I was on the 5A firmware)

mcheah, 2020-03-20:
There are certain heating conditions that can trigger thermal runaway due to the firmware only firing one heater or the other at a time (for example starting both heaters, then manually changing the bed temp afterwards).  So if it was triggering thermal runaway (which logs from pronterface would tell us), that would make sense on the 5A firmware.  Regardless, I'm glad you've resolved your issue.