mcheah/Marlin4MPMD #60: when i start a print it does not caribrate/ auto level it just starts printing

Krishpannze, 2020-02-25:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Which g-codes can cause this issue?
2. Are you printing from USB or the SD card?
3. Are you using the LCD display to start prints or are you using commands from the USB?
4. What OS are you using?
5. What software are you using?(Cura, Pronterface, Octoprint, Repetier...)
6. Did this work on the stock Malyan firmware?
7. What error message is reported?  Are the LED lights blinkly quickly
8. Post the full-code and status log, if you can.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Software: [Cura, Pronterface, Octoprint, Repetier...]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

mcheah, 2020-02-25:
Your start gcode looks fine to me.  What's the specific problem you're experiencing?

Krishpannze, 2020-02-25:
> Your start gcode looks fine to me. What's the specific problem you're experiencing?

The prints are not sticking to the bed the first layer is getting messed up.
im also pretty new to all of this.

Krishpannze, 2020-02-25:
> Your start gcode looks fine to me. What's the specific problem you're experiencing?

Also when i try to auto level it wont let me save to eeprom when i send a m500 to save it just says sending:M500

mcheah, 2020-02-25:
Here's a couple guides to look through to get you started:

https://www.reddit.com/r/mpminidelta/comments/bzm1s2/updated_mpmd_calibration_guide_and_faq/

https://github.com/mcheah/Marlin4MPMD/wiki/Calibration

Play around with M851 or G29 Z to adjust the first layer height.

Krishpannze, 2020-02-25:
I am not able to save anything to the eeprom though

mcheah, 2020-02-25:
Can you provide a log to demonstrate this?  The firmware doesn't give any message to indicate it is saving settings, but it should save things correctly with M500.  Post your M503?

Krishpannze, 2020-02-26:
my M503 is 

SENDING:M503
echo:Steps per unit:
echo:  M92 X114.29 Y114.29 Z114.29 E97.00
echo:Maximum feedrates (mm/s):
echo:  M203 X150.00 Y150.00 Z150.00 E50.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X800 Y800 Z800 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z20.00 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Endstop adjustment (mm):
echo:  M666 X0.00 Y0.00 Z0.00
echo:Delta settings: L=diagonal_rod, R=radius, S=segments_per_second, ABC=diagonal_rod_trim_tower[123]
                DEF=diagonal_rad_trim_tower[123],XYZ=delta_tower_angle_trim[123]
echo:  M665 L120.80 H124.03 R61.70 S200.00 A0.00 B0.00 C0.00 D0.00 E0.00 F0.00 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P45.00 I0.07 D1000.00
echo:  M304 P80.00 I12.71 D300.00
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-0.60

and the m421 is

SENDING:M421
Grid spacing X:15.00 Grid spacing Y:15.00
0.42 0.42 0.66 0.46 0.48 0.42 0.42
0.42 0.72 0.57 0.39 0.33 0.42 0.42
0.78 0.54 0.36 0.22 0.18 0.26 0.24
0.60 0.39 0.25 0.00 -0.09 -0.01 0.03
0.46 0.22 0.04 -0.04 -0.12 -0.21 -0.25
0.42 0.13 -0.03 -0.15 -0.25 -0.30 0.42
0.42 0.42 -0.02 -0.24 -0.31 0.42 0.42

Krishpannze, 2020-02-26:
[image]

This is what the m500 to save things shows when I type it. It is not saving my g29 I think

Krishpannze, 2020-02-26:
i Just reflashed the software and now m500 is working

Krishpannze, 2020-02-26:
but i was wondering on how to change the z offset