mcheah/Marlin4MPMD #61: Printer reseting/restarting

Krishpannze, 2020-03-04:
Hi when i try to print this a couple minutes into the print the printer resets / reboots and i am not sure why. when it resets it shows the version number on the bottom of the screen and it stays there. and i have not been able to get any successful prints

Krishpannze, 2020-03-04:
the temp settings are 250/75

Krishpannze, 2020-03-04:
I just checked in pronter face and this is what i got 
Error:Heating failed, system stopped! Heater_ID: 0
[ERROR] Error:Heating failed, system stopped! Heater_ID: 0

Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

mcheah, 2020-03-04:
Can you provide a full pronterface log as well as a picture of the temperature plot?  Please enable "debug communications" in pronterface and issue M111 S17 before heating.

Krishpannze, 2020-03-05:
I just did a m421 and restored everything to defaults and now it's fine thanks for the help.

Krishpannze, 2020-03-05:
Yes It is still having the problem so i will send the things tommorow

ben-gratias, 2020-03-17:
Hi, I am having the same issue I believe. I have attached a couple screenshots showing the Pronterface output from the printer when the error occurs. The printer repeatedly tried to reach temp but starts oscillating in the 180s without hitting the target. This appears to be a behavior to not overshoot temperature by shutting off the heating element just before it hits the target but it doesn't make it and gets stuck in an endless loop.

Console output:
[image]

Temp Graphs:

Begins fine:
[image]

Plateaus in the 180s:
[image]

(Am I correct in thinking that the 127 in this line is power level for hot end? If so it does definitely switch to 0 before temp is reached.)
[image]

mcheah, 2020-03-17:
Hi,

Thanks for providing the logs and the chart, this is very useful and actionable information.
From the pronterface logs you can clearly see that thermal runaway is getting triggered from heater 0 (the main extruder).

From your temp graphs I can see that neither the bed nor the extruder ever reach a settled temp, which in turn triggers a thermal runaway error and shuts down the printer.  This is intended behavior because the printer knows something is wrong because it's been trying to heat for several minutes and it still can't get to temp.  Again, this is working as expected.  

You are correct the `@:127 B@:52` refers to the heater duty cycle at that particular moment, 0=0% duty cycle, 127 =100% duty cycle.

Consider a situation where the thermistor falls out of the heat block, the printer thinks everything is at room temp and turns on the heater harder and harder to compensate.  In reality the heat block is much hotter and starts heating to dangerous critical temperatures.  The printer has no idea and cannot fix the situation.  You can see why this is dangerous.  To combat this, the printer has two checks to prevent this particular error.  1.  It checks that the heater rises by a certain number of degrees per second while heating.  If temperature does not rise it triggers an error.  2. Once temperature has settled it checks that the temperature is stable +/- some threshold around the set point.  If the temperature swings too far outside of a range it triggers an error.

From your chart it seems to me that you are likely triggering the first one.  The most likely culprit is tha the PID values are mis-tuned.  These should not be copied from your stock firmware as the time interval they are calculated over is different.  The first thing you should try is issue M502 to reset PID to default values and repeat your experiment.  There are other possibilities but let's start there.

Some additional questions or suggestions that can help me understand what's causing your issue
1.  Provide your full M503 details
2.  Are you running a stock system (hot-end, power supply, fan/part cooling, etc?)
3.  Are you running the 5A or 10A firmware?
4.  Can you enable "debug communications" and log the entire temperature process, and then attach it as a .txt?  This will tell me information about how the printer is making decisions while heating.
5.  Do you remember the original firmware version?  Also can you tell me approximately when you purchased this printer?  There have been many hardware revisions and I'd like to know if this is a problem specific to a variant.

ben-gratias, 2020-03-17:
Thanks for your quick response. I am at work currently but will send more detailed information when I get home. I will also try issuing the M502 and see if that fixes the issue. I'll do the debug communications both before and after the M502 to get as much info as possible.

1) Will send later
2) Hot-End is stock, but I am running upgraded 10A PSU, and have added an additional cooling fan (Just tied in to the same control for hot-end fan).
3) 10A firmware
4) Will send later
5) 90% sure it was V44.161.3 but I believe I made note of this and will update later if not correct.

ben-gratias, 2020-03-18:
1)
[image]

4)
[troubleshooting.txt](https://github.com/mcheah/Marlin4MPMD/files/4346794/troubleshooting.txt)

ben-gratias, 2020-03-18:
Oh I forgot to mention, this printer was purchased sometime around December 2019

PurpleHullPeas, 2020-03-18:
@ben-gratias You're using the PID values tuned by Dennis Brown specifically for the hacky stock firmware. Marlin4MPMD handles PID properly, so you will need to perform a proper PID autotune.

1. Delete any M301 and M304 values out of your Start Gcode.
2. Run a proper PID autotune and save it. e.g., 
> M303 C10 E0 S190 U1
> M500
3. Run M503 again to make sure that they actually saved.

ben-gratias, 2020-03-18:
@PurpleHullPeas thanks, will try later today

ben-gratias, 2020-03-20:
Everything seems fine now, thanks all for the help. I will update if any of the issues come back.