mcheah/Marlin4MPMD #62: New Hotend with different Thermistor, breaking both temperatures when adjusting config

ppetermann, 2020-03-04:
**Describe the bug**
So, I've replaced the stock Hotend with an E3D Lite6, which comes with a cartridge thermistor, that according to E3D works with the table corresponding with 5.

So I've edited the config accordingly and built Marlin4MPMD, but when I boot, and look at the temperature readout in the preheating screen, it now shows 0 for both thermistors (i've only changed from 75 to 5 on the hotend one).
If i go back to 75, the bed one is correct, and the hotend one shows a number that might be closer to correct than 0, but probably isn't accurate.

setting both to 5 (which should make the bed read wrong, which i could live better with than with the hotend being wrong) also results in a 0/0 reading.

**To Reproduce**
Steps to reproduce the behavior:
1. define TEMP_SENSOR_0 5
2. build for release, put firmware.bin on sdcard, boot.
3. go to heating menu -> reads 0 / 0 

**Expected behavior**
should read current temperature (room temperature slightly above 20C) for both

mcheah, 2020-03-04:
Try the [thermistor_5](https://github.com/mcheah/Marlin4MPMD/tree/thermistor_5) branch and see if that works for you.

ppetermann, 2020-03-05:
the x4 multiplier in there fixed it.

closed 2020-03-05 by ppetermann